jotego/jtframe #29: SDRAM Fast Output usage

gyurco, 2020-08-14:
This expression:
```
assign SDRAM_DQ = write_cycle ? dq_out : ( hold_bus ? 16'h0 : 16'hzzzz);
```
inserts extra combinatorial logic between dq_out and SDRAM_DQ, thus prevent the usage of Fast Output registers.
I would suggest to directly register SDRAM_DQ in ST_TWO:

```
        SDRAM_DQ <= 16'hZZZZ; // default

        if ( cnt_state == ST_TWO ) begin // set read/write
            // sdram_ack     <= 1'b0;
            SDRAM_A[12:9] <= 4'b0010; // auto precharge;
            SDRAM_A[ 8:0] <= col_addr;
            {SDRAM_DQMH, SDRAM_DQML } <= write_cycle ?.
                ( downloading ? prog_mask : sdram_wrmask )
                : 2'b00; // reads always take the two bytes in
            SDRAM_CMD   <= write_cycle ? CMD_WRITE :
                refresh_cycle ? CMD_NOP : CMD_READ;
            dq_rdy      <= 1'b0;
            if (write_cycle) SDRAM_DQ <= dq_out;
        end
```

SDRAM_DQ should be registered together with the write command. I also don't understand the effect of the bus hold on MiSTer, thus I won't send a patch. But the above method should be the correct one.

jotego, 2020-08-14:
That makes a lot of sense. In my initial tests with Z logic I only found it to work outside of an always statement but I'm going to try what you suggest.

jotego, 2020-08-14:
I have made that change in this commit:

https://github.com/jotego/jtframe/commit/d59415cc767e613124ddf9c3c6e48b7c239369ba

I'm compiling now in MiSTer which was more problematic about SDRAM_DQ timings than MiST, where things normally go smooth.

Thanks a lot for this very needed suggestion.

gyurco, 2020-08-14:
Outputting active low level in a read cycle doesn't look good to me. I'm not familiar with the input/output stages of the FPGA and SDRAM, but if the bus is actively pulled down by the FPGA, how the SDRAM supposed to also output some meaningful signals?
With a standard totem-pole output, if the transistor which pulls down the output to zero is active, then an incoming high level almost short-circuited to the ground.

jotego, 2020-08-14:
Unless I've made a mistake with this edit, the idea is to hold the bus at zero and release it just before the SDRAM drives it. That makes the voltage transitions more consistent. It works very well for MiSTer where the memory electrical connection is poor.
The FPGA and the SDRAM should not drive the bus at the same time -unless I've introduced a bug in the last commit

gyurco, 2020-08-14:
Then hold_en shouldn't be hold_bus here?
```
SDRAM_DQ <= hold_en ? 16'h0 : 16'hzzzz;
```

jotego, 2020-08-14:
You see. I cannot work on Fridays! Haha
Yes, I've changed to:
`SDRAM_DQ <= hold_en && hold_bus ? 16'h0 : 16'hzzzz;`
I'm compiling again now. Thank you.

gyurco, 2020-08-14:
I wonder if it will be possible to push the output enables of SDRAM_DQ to the Fast Output Enable with this logic :)

jotego, 2020-08-14:
hold_en is a fixed signal during compilation. So only hold_bus drives it.

gyurco, 2020-08-14:
And the state machine logic...
As I suspected, it's not a Fast Output Enable now:
```
+--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
; Bidir Pins
+--------------+-------+----------+--------------+--------------+--------------+-----------------------+--------------------+--------+----------------+-----------------+--------------------
; Name         ; Pin # ; I/O Bank ; X coordinate ; Y coordinate ; Z coordinate ; Combinational Fan-Out ; Registered Fan-Out ; Global ; Input Register ; Output Register ; Output Enable Regis
+--------------+-------+----------+--------------+--------------+--------------+-----------------------+--------------------+--------+----------------+-----------------+--------------------
; SDRAM_DQ[0]  ; 83    ; 5        ; 53           ; 11           ; 0            ; 0                     ; 13                 ; no     ; yes            ; yes             ; no
```
Maybe the whole problem you're trying to solve with the bus hold comes from this? (It wasn't a Fast Output, nor Fast Output Enable before).

gyurco, 2020-08-14:
Ah, this one has to be inserted to mist.qsf (and sidi.qsf, too):
```
set_instance_assignment -name FAST_OUTPUT_ENABLE_REGISTER ON -to SDRAM_DQ[*]
```

jotego, 2020-08-15:
Thanks a lot for your help. I have extended this to MiSTer too and these signals are now using registers at the pads for all systems.
:-D

closed 2020-08-15 by jotego