jotego/jtframe #35: OSD and control rotation behavior

gyurco, 2020-09-05:
It seems to be random which core actually rotate the OSD with rotate controls and which is not.
Also control rotation doesn't honor flip screen, for example in 1942, Rotate Controls Yes, then the controls are OK when Flip is Off, but not reversed when flip is On.
(And in 1942, the OSD stays in landscape mode even in a portrait usage)
For this purpose, I've written an arcade_controls module, which handles control rotation in every possible case:
https://github.com/mist-devel/mist-modules/blob/master/arcade_inputs.v
Maybe something similar can be added to jtframe, too.

gyurco, 2020-09-05:
I see that MiST always get a landscape OSD (why?) and apply_rotation in jtframe_board can rotate the controls only in one direction. Maybe a flip option for this function could be used.

jotego, 2020-09-06:
The OSD rotation comes from that extra byte in ROM with index 1. There needs to be an ARC file to make it work.

jotego, 2020-09-06:
1942 seems to rotate the OSD ok, including the flip (after resetting the core). Maybe some ARC files are missing the extra byte.

gyurco, 2020-09-06:
Hmm, there must be some problem with the MRA utility, since MOD=0 in 1942.ARC

gyurco, 2020-09-06:
The problem is that the MRA contains:
```
<rom index="1">01</rom>
```
But should contain a part tag:
```
<rom index="1"><part>01</part></rom>
```

gyurco, 2020-09-06:
And 1943, Gunsmoke, etc.. misses the whole rom index="1"

closed 2020-09-06 by gyurco