jotego/jtframe #42: Two SDRAM issues with the new controller

gyurco, 2021-01-04:
The new SDRAM controller has an issue with register packing:
    Warning (176279): Can't pack register node "jtframe_mist:u_frame|jtframe_board:u_board|jtframe_sdram_bank:u_sdram|jtframe_sdram_bank_core:u_core|sdram_a[12]" into I/O pin "SDRAM_A[12]".
 The node "jtframe_mist:u_frame|jtframe_board:u_board|jtframe_sdram_bank:u_sdram|jtframe_sdram_bank_core:u_core|sdram_a[12]" cannot simultaneously use clear and load signals.

The set_global_assignment -name FORCE_SYNCH_CLEAR ON doesn't help on it any more.

Also:
set_instance_assignment -name FAST_INPUT_REGISTER ON -to SDRAM_DQ[*]
is no longer in the qsf, lost in time...

gyurco, 2021-01-04:
This one fixes this:
```
diff --git a/hdl/sdram/jtframe_sdram_bank_core.v b/hdl/sdram/jtframe_sdram_bank_core.v
index d7afa29..8fd27a7 100644
--- a/hdl/sdram/jtframe_sdram_bank_core.v
+++ b/hdl/sdram/jtframe_sdram_bank_core.v
@@ -225,6 +225,7 @@ always @(posedge clk, posedge rst) begin
             init_cmd <= CMD_NOP;
             cmd      <= init_cmd;
         end else begin
+            sdram_a <= 0;
             if(!init_st[2]) init_st <= init_st+3'd1;
             case(init_st)
                 3'd0: begin
@@ -291,8 +292,10 @@ always @(posedge clk, posedge rst) begin
             if( ADQM )
                 // A12 and A11 used as mask in MiSTer 128MB module
                 sdram_a[12:11] <= wrtng ? wrmask : 2'b00;
-            else
+            else begin
                 dqm            <= wrtng ? wrmask : 2'b00;
+                sdram_a[12:11] <= 0;
+            end
             if( wrtng ) begin
                 dq_pad <= din;
                 cmd    <= CMD_WRITE;
```

jotego, 2021-01-05:
I do not understand why the change makes a difference. Do you understand why?

gyurco, 2021-01-05:
Not really, it has something how the LAB's clear and load signals are used by the synthesizer.

jotego, 2021-01-05:
I think the only stable solution is to instantiate the IO primitives directly, so the synthesizer stops messing with our lives. But it breaks code portability.

gyurco, 2021-01-05:
But setting to 0 the otherwise not-relevant bits doesn't hurt. The sdram_a[12:11] <= 0 won't happen on MiSTer's sensitive sdram. The other one only done during init.

jotego, 2021-01-05:
I have committed the change. By looking at the results, it's funny that SDRAM_DQ do not appear to be using the pad register for input, only for output.

gyurco, 2021-01-05:
Yes, because:

Also:
set_instance_assignment -name FAST_INPUT_REGISTER ON -to SDRAM_DQ[*]
is no longer in the qsf, lost in time...

closed 2021-01-09 by jotego