jotego/jtframe #57: mist_io.sdc has wrong clock for the SDRAM

gyurco, 2021-12-25:
u_clocks prefix is missing, thus the SDRAM signals are not constrained at the end. Easy fix:

[code]
@@ -27,7 +27,7 @@ create_clock -name {SPI_SCK}  -period 41.666 -waveform { 20.8 41.666 } [get_port
 derive_pll_clocks -create_base_clocks
 
 create_generated_clock -name SDRAM_CLK -source \
-    [get_pins {u_pll_game|altpll_component|auto_generated|pll1|clk[2]}] \
+    [get_pins {u_clocks|u_pll_game|altpll_component|auto_generated|pll1|clk[2]}] \
     -divide_by 1 \
     [get_ports SDRAM_CLK]
[/code]

PS: Merry Christmas!

jotego, 2021-12-25:
Blast! I introduced this error when I moved the PLLs into the clocks module. Thanks a lot for telling me. Fixed in 667395f

closed 2021-12-25 by jotego

gyurco, 2021-12-25:
Forgot to mention: it also affects the PLL location, too.
https://github.com/jotego/jtframe/blob/master/target/mist/mist.qsf#L155
Should be
```
set_location_assignment PLL_1 -to "jtframe_mist_clocks:u_clocks|jtframe_pll0:u_pll_game|altpll:altpll_component"
```