jotego/jtframe #71: Feature request: Support for HDMI pixel repetition

Toryalai1, 2022-11-18:
Hi Jotego,
Current MiSTer firmware has support for HDMI pixel repetition I wonder if you could consider adding this to your framework/cores as well. :)

jotego, 2022-11-21:
Could you elaborate a bit more about it? What is this? Why is this useful? Is this something that just come with the MiSTer sys now?

jotego, 2022-11-23:
Information from Toya:

Documentation: https://mister-devel.github.io/MkDocs_MiSTer/advanced/videomodes/?h=pixel#additional-flags
Original PR: https://github.com/MiSTer-devel/Template_MiSTer/pull/64/files

jotego, 2022-11-23:
Feature added in commit 0d3ae6e. I will be recompiling and uploading the new cores during the following days (it takes some a long time to compile and test before breaking everybody's systems, and I'll probably end up breaking something anyway!)

jotego, 2022-11-23:
BTW, thanks for bringing this topic up and thanks to Toya for checking a couple of test cases.

closed 2022-11-23 by jotego