jotego/jtframe #104: Spinner not working correctly

sonik-br, 2023-02-09:
Hi,

I'm trying to get spinner working on the 3 (currently) supported games.
Tests where made with different hardware and the problem is exactly the same with all.

Forgotten Worlds:
Cant rotate CCW. CW works as expected.
Even when not physically rotating the spinner, the character on screen have a very slow CW rotation.

Eco Fighters and Puzz Loop 2:
Can rotate both ways. CW works as expected. CCW sometimes locks, does not move at all.

Hardware tested:
Namco JogCon with arduino adapter.
https://github.com/MiSTer-devel/Retro-Controllers-USB-MiSTer/tree/master/JogConUSB

mr.Spinner - rotary encoder wired to arduino.
https://github.com/MiSTer-devel/Retro-Controllers-USB-MiSTer/tree/master/PaddleTwoControllersUSB

Those two devices are seen by mister as a spinner. Works as expected on the Input Tester core, arkanoid core, nes core, sms core...

Also tested mouse by configuring mister to use all mouse as a spinner ([ini setting](https://github.com/MiSTer-devel/Main_MiSTer/blob/master/MiSTer.ini#L205))
Tested a mini keyboard with a track pad.
A simple dell mouse.
A PS/2 trackball with a simple ps/2 to usb adapter.

Mouse as spinner also works across mister cores and input tester core.

jotego, 2023-02-09:
I need a device I can buy. It looks like those are devices to build at home.

sonik-br, 2023-02-09:
Yes, they need custom adapters.
Mouse as spinner is something easy to try with any mouse. Just need a setting on the ini. I used the id `FFFF` to force all mouse to be used as spinner.

ShinKaz, 2023-02-09:
> I need a device I can buy. It looks like those are devices to build at home.

what about the EgretIImini spinner/trackball ? :)

aswgit, 2023-02-10:
I'm using both a home built Mr.Spinner and this device:  https://www.amazon.com/dp/B09DNYJB68

Forgotten Worlds will not recognize CCW.

Puzz Loop 2 and Eco Fighters will eventually stop rotating CCW.  I can reproduce this 100% by rotating it CW 4 full revolutions, then rotating it CCW 4 full revolutions, until it stops moving.

In Eco Fighters service mode, the input display jumps between 0-1 when rotating CCW if input sensitivity is set to lowest.  It stays at 1 if it's set to highest.  This is visible in gameplay as well with the pod spinning faster CW than CCW at lowest.

PapiFed, 2023-02-10:
@jotego I can confirm the same problems about CCW getting stuck with the spinner from Ricardo so you should be able to reproduce on your end.
Same device works fine with Arkanoid arcade and Pop'n Bouce + Hypernoid on Neo Geo.

closed 2023-02-26 by jotego

jotego, 2023-02-26:
I found the problem. New binaries in https://github.com/jotego/jtbin/commit/5e4926c867e160d91fb85ceba16581ac104f0747.

commit 025a581f (2023-02-26): Fix #104