jotego/jtframe #108: Design a new development FPGA

jotego, 2023-03-11:


somhi, 2023-04-08:
Some ideas for helping you develop faster and better. :)

Platforms:

* MiST compatible platform with a bigger FPGA board than MiST or SiDi. E.g. take a de10-nano and make it MiST compatible.  https://atari-forum.com/viewtopic.php?p=445410#p445410,    https://github.com/somhi/MiSTdon   (I can send you a  PCB of the MiSTdon)

* DeMiSTify platform with a bigger FPGA. No external CPU, but a synthetic CPU instead is used to aid the core for OSD options, loading rom files, HD images, arc files, ....    See current targets https://github.com/robinsonb5/DeMiSTify/tree/main/Board      Adding new targets is easy and de10-nano will be added someday soon

  Does not have all features like MiST because this platform is focussed on low FPGA resources and all the firmware is loaded in BRAM, but that could be expanded.

​		Alastair M. Robinson is at reach, and he also wants to integrate DeMiSTify someday in jtframe.   

Possible FPGAs to be used with platforms above:

* DE10-nano
* Cyclone 10 LP Refkit 10CL055  https://shop.trenz-electronic.de/en/TEI0009-02-055-8CA-Cyclone-10-LP-RefKit-10CL055-Development-Board-32-MByte-SDRAM-16-MByte-Flash   easily expandable with pmods dongles.
* QMTECH Cyclone IV 150 kLE https://es.aliexpress.com/item/1005004065727282.html
* QMTECH Cyclone V 77 kLE   (I see no stock ATM, but some people have bought it recently)

Those QMTECH will need and addon, the official one from QMTech or like those of Antonio Villena with SRAM memory.

jotego, 2023-05-01:
I have got [one of these units](https://shop.trenz-electronic.de/en/TEI0022-03-DataStorm-DAQ-M-Board-FMC-Carrier-for-M-Series-Precision-Converters), which looks promising. I'm going to explore that option for a while.

| Feature        | MiSTer       | DataStorm         |
|:---------------|:-------------|:------------------|
| Cyclone V FPGA | 5CSEBA6U23I7 | 5CSEMA5F31 C8N    |
| Logic Elements | 110k         | 267k              |
| DSP blocks     | 360          | 1140              |
| Transceivers   | 4            | 12                |
| IO pins        | 330          | 484               |
| Memory Mbits   | 4.9          | 14.4              |
| Package        | UBGA-484     | FBGA-896          |
| DDR            | 1 (ARM)      | 1 (ARM), 1 (FPGA) |
| SDRAM          | Yes          | No                |
| Video output   | HDMI         | HDMI ADV7511      |
| USB            | Yes          | Yes x 4           |
| DIP switches   |              | 3 x 4 bits        |
| micro SD port  | Yes          | Yes               |
| Flash RAM      |              | 32 MB x 2         |
| HF connectors  | No           | Yes               |

closed 2023-05-01 by jotego